matsstaff/stc1000p #52: Suspicious RANGE() function

ArtMilburn, 2015-03-10:
Hello,

I was really looking forward to using this program on my STC-1000, but unfortunately I got a version with a Hollman processor on it rather than the Microchip part.

However, in looking through the code briefly, I found a potential problem that you might want to take a look at:

static int RANGE(int x, int min, int max){
	if(x>max)
		return min;
	if(x<min)
		return max;
	return x;
}

If x is greater than the max, then the min is returned!?  Seems like the max should be returned.  Same kinda deal for x less than min.

matsstaff, 2015-03-10:
Hi!
Thanks for the report! It is nice to know someone gives a hoot :)
However, it is really the name of the function that is misleading. It is used to limit the values that you enter with the keypad. I personally (as I think a lot of others would also), prefer it to 'roll over' when max of min is reached, instead of going all the way back in the other direction. 
But you are correct in that it is not really a range. I just don't really have a better name...

Cheers!

ArtMilburn, 2015-03-10:
OK, I get it.  It's a rollover, not a clipper.

Well, no harm, no foul.

Keep on brewing,

Art

closed 2015-03-10 by ArtMilburn

ArtMilburn, 2015-03-24:
Mats,

Just wanted to let you know that I ended up buying an STC-1000+ with dual
probes from Will at Black Box Brew.

Thanks again for your hard work,

Art