matsstaff/stc1000p #67: Buttons reversed

hebs, 2015-10-25:
I purchased the stc1000 (a400p) on amazon and I was able to flash it, etc... However it appears that my buttons are reversed (see picture below). The arrow buttons are on the left and the power/esc and selection buttons are on the right. 

https://lh3.googleusercontent.com/-oYIOGAyPlZo/ViiJorO-r3I/AAAAAAAAX2g/JoTXVy4AX1Q/s912-Ic42/DSC09396.JPG

This isnt really a "defect" in you firmware as much it is as a quality/consistency issue with the hardware itself, since most of the pictures I see of the stc1000 have the buttons oriented differently than mine... but would it be possible to get you to add a fw configuration option to re-map the buttons?

I purchased this stc1000 on amazon after someone else in an HBT thread mentioned that they were a400p's. Once more people catch on to your firmware, and the fact that there is an a400p available via prime on amazon, these are going to be selling like hotcakes. If all of the ones sold in the above amazon link are a400p's, there will be lots and lots of people wanting this feature. You would think it would be easy to get used to having them reversed, but its damn near like driving a car with the steering reversed.

(amazon link: http://www.amazon.com/gp/product/B008KVCPH2?psc=1&redirect=true&ref_=oh_aui_detailpage_o03_s00 )

stoey, 2015-10-25:
I think the rubber button caps just need to be rotated 180 degrees. The power symbol is upside down.

matsstaff, 2015-10-25:
Agreed. This is not a software issue. Try to open the unit and rotate the keys.

closed 2015-10-25 by matsstaff