matsstaff/stc1000p #68: New A400-p PCB from ALiExpress "U-Control"

hidara2000, 2015-11-29:
Hi Matsstaff,
I ordered a couple of STC1000s, with the A400-P chip, from U-Control. On opening the case I noticed that the programming head was missing on the main PCB. I did, however, notice that there is an ISP near the LCD. I wonder is it possible to use this to reprogram the device and is there anyway to detect what each connection is?
Last week I asked "U-Control" if they would confirm if these points correspond to 5V,GND,ICSPCLK(9) etc. but I have had no reply yet.

http://www.aliexpress.com/snapshot/7110112707.html?orderId=71007002417744
See attached pics
[image]
[image]
[image]
[image]

Cheers for any advice  
D

matsstaff, 2015-11-29:
First check to see that the MCU is the same, PIC16F1828. From the pictures, it seems like it cpuls be the same (same number of pins), but if you can, try to take a close up photo of the chip and see if you can read it out.
If the MCU is the same, then it seems plausible the ISP header is indeed what we need, it is worrying that there is an extra pin, but it might just be to conform with the pickit programmer that normally has 6 pins. If so, I would think that the connections would be (from your picture)
1 -> nMCLR
2 -> VDD
3 -> GND
4 -> ICSPDAT
5 -> ICSPCLK
The only way to be sure in advance would be to trace or measure the connections to the MCU, but if you can measure +5 volt between pin 2 and 3 then, I would think you're good.

Very interesting discovery! Thanks for sharing and please let me know how it turns out!!!

Cheers!
//mats

hidara2000, 2015-11-30:
Hi Mats.
Cheers for the reply. I only have a basic knowledge in electronics but I took my multimeter to those pins and this is what I found.

1-2   2.265v
1-3   5.000v
1-4   0
1-5   0
1-6   5.000v
2-3   2.727v
2-4   2.270v
2-5   2.269v
2-6   2.730v 
3-4   5.000v
3-5   5.000v
3-6   0
4-5   0
4-6   5.000v
5-6   5.000v

I've also added some extra photos. Any help would be appreciated :) I'm not to worried if I brick it, as the only reason I got it was to reprogram it. The site did say that it was the A400-P chip but I cant make out the numbers :/

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Cheers Again
D.

matsstaff, 2015-12-01:
You know what? I don't think that really is an A400_P unit at all.

* From the picture, it is clear that pin 5 is GND. That doesn't make sense.
* It is clear that the LED/MCU PCB would not fit an A400_P base PCB.
* There are no markings on the base PCB saying that it is A400_P
* I can't read the markings on the MCU, but compared to my own, it does not really look like the Microchip 'M' on there. Probably not a PIC MCU then.
* The base PCB looks like one of the clones that have been discussed in the homebrewtalk thread.

Did you *really* get this from U-control? No chance of a mixup? I find it hard to believe they would send a unit not built by themselves. If so, then I'd have a word them. This is pretty clearly not an A400_P unit.

Sandedk, 2015-12-14:
I also bought 2 from U-Control. Not A400_P
http://www.aliexpress.com/item/temperature-controller-STC-1000/2004466302.html

mattrox, 2015-12-19:
Apparently a mix-up.

The 2 I recently got are the correct version.

hidara2000, 2015-12-20:
I have spoken to the supplier and they are going to send me the correct version. I think, moving forward, that I might confirm with Ucontrol that they will send the correct version before I order it. Cheers for the advice matstaff!
D

drops77, 2015-12-21:
I contacted them when I read this, seems their was a mix up in the warehouse and has been resolved

http://postimg.org/image/k62kebvar/

matsstaff, 2015-12-23:
Good to know! I hope this matter is sorted then and won't happen again.
For the record, the contact I've had with U-control has been nothing but good, but I have had mixups happen before (got 110v versions instead of 220v), but they were helpful in sorting it out.
Also, one needs to keep in mind (as always when ordering from Asia), that they offer the items at a slim margin and shipping worldwide is not free (even if you might be led to believe that). To expect the same level of service as when shopping locally (at significant markup), is not fair. However, U-control has in my experience been very nice to deal with.

closed 2015-12-23 by matsstaff

hidara2000, 2015-12-29:
Following from matstaff's last comment, I second that. The contact I had was most helpful and the correct items arrived in the post today. I order quite an amount of products overseas and sometimes all suppliers do make honest mistakes. I like to judge them on how they resolve these issues and in this case U-control have been top notch.
Thanks again matstaff for your advice and for sharing your work.

owenduffy, 2019-02-07:
I also purchased on from the U-CONTROL store on Aliexpress... and the classic bait and switch trick was performed by the Chinese Cheats, the controller was expensive and was not an A400_P.