matsstaff/stc1000p #70: Get Device ID error message

CNevay, 2015-12-22:
Hey Matt, 

I'm trying to upload my 2nd controller but with a new Uno clone that I've never used before. When trying to upload the most recent picprog.ino to the uno, I am continually getting the following error message:

[image]

Have you ever come across this before? 

Thank you again for the software, although I didn't buy a Black Box i still made a donation to the Dana-Farber Cancer Institute in your name, as a gesture of thanks. 

Thanks, 

Cam

CNevay, 2015-12-22:
I uninstalled IDE 1.6.7, installed 1.6.3 and was able to upload the .ino file no problem. Please consider this closed.

matsstaff, 2015-12-23:
Cool! Glad you got it working!
And big thanks for the donation! Very nice of you!
Cheers!
//mats

closed 2015-12-23 by matsstaff