matsstaff/stc1000p #86: Improve 'button up' information in OVBSC

matsstaff, 2016-01-31:
See if it would be possible to show SP even if in constant power mode when running program.

--
I have a few suggestions that I think will contribute to ease of use of the software.
They are all related to what's on display when running the program.
When first running program and waiting for St to be reached, a press on Up reveals Output Effect (in my case with one heating element: 100.
I think it would be more useful to present St temperature setpoint instead of output. By that I get a verification of what temperature the system is heading at, and can approximate what time is left until there.

When into the different Mashing steps, a press on the arrows displays actual mashing steps and setpoints. Would it be possible to display remaining time instead for example of the name of the step?
For example, a standard 65*C and 60 min would then display 65 when pressing Up and remaining time when pressing DOWN. By that you can figure out how long time is left until next mashing-step and until time to do lautering etc.

The same goes for several other steps, where Output Effect is presented when pressing UP.
--

ep98juma, 2016-02-07:
Hello again! 
Today I have found out that "button down" toggles name of parameter and remaing time when press and hold, just as you replied! GREAT!
I suggest that "button down" function first presents remaining time and then toggles to the name of the step after say 1 sec. By this, the most useful information is easy accessed. (button up = temperature step in mashing and button down= remainig time in that step). 

This software is anyway awsome and makes brewing day a charm!

PS Is there space enough for the communication code in the OVBSC version? It would be handy in the fduture to be able for example to edit the parameters on the phone (by use of and arduino with wifi / bluetoth or similar) and then just communicate the new values to the STC. 

Hälsningar Magnus

matsstaff, 2016-02-08:
Ok, valid point. I do think that might just be down to personal preference, but it shouldn't be to hard to change and I can agree with your reasoning. 
Gut feeling says communication stuff probably could fit in OVBSC, but then you'd have to choose to between pump control or com. It uses the same pin (the only one that is available). 
Might not be a bad idea anyway.