matsstaff/stc1000p #90: OVBSC: not entering thermostat mode when strike temperature reached

karlp, 2016-02-28:
I'm going to have run some more tests on this, but it doesn't seem to be switching to thermostat mode when strike temps are reached.  Today I came back late, the alarm was beeping, but the temp was up at 75C, (strike was meant to be 69),  I don't think my P0 output ratio was too high, it was only 30, and later during first mash, it wasn't even holding heat and I had to raise it to 45%.

matsstaff, 2016-03-31:
Do you have any additional information on this issue?
Have you checked the APF flags are set correctly? 
I don't get a ton of feedback from users, but at least a few have reported OVBSC working as expected.

karlp, 2016-03-31:
will be brewing again tomorrw and will recheck, I realise it's a kinda vague report :(

karlp, 2016-08-12:
As far as I can tell, it doesn't switch off the S0 output when St is reached, only when the alarm is acknowledged and you enter the "pause" mode for dough in.

matsstaff, 2016-08-12:
That is excellent information! Makes it a lot easier to troubleshoot. I'll look into it. Thanks!

matsstaff, 2016-08-13:
Are you running the latest code (that is current from master branch)?
Because, I've been mulling over the piece of code, knowing what to look for, and it *really* looks like it should switch to thermostat mode with PO output right after St is reached and before the alarm/pause.
Cheers!

karlp, 2016-08-13:
I'll have to try again, going to have to get on it more anyway, if I ever want to nail down this mash temp

karlp, 2016-09-25:
After brewing again tonight, I believe this is behaving "as designed" but it's very susceptible to the tuning of the P0 factor :|  Setting S0 to a low enough ratio to not overshoot much is just pointlessly slow, so it overshoots there (no circulation to keep the thermoeter stable) and then P0 has a tendendecy to either heat too much or not keep heat.  I believe my P0 setting was simply too high in the past.

closed 2016-09-25 by karlp