matsstaff/stc1000p #92: OVBSC Whirlpool Step

flynut, 2016-06-03:
Thanks Mat for all of your great work.  I thoroughly enjoy my brewing day now!!
I think a nice enhancement to the OVBSC code would be to add a whirlpool step after the cooling step that only runs just the pump for a timed interval.  Placement of this step after the cooling step would allow brewers to cool the wort below the hop isomerization temperature and then whirlpool their flame-out hops.

matsstaff, 2016-06-08:
Cooling (and whirlpooling, hopstand and whatnot) has not been part of the OVBSC 'program' as it is out of its control. However, you are not first in wanting such features.
One problem would be the **APF** parameter, that is full. It could probably be solved (splitting it to two parameter like **AP1** and **AP2**).
I'll think about it, maybe coolingand whirlpool/hopstand could be worked in, but I can't make any promises.

Kalsta, 2016-09-02:
How about just adding a couple more hop alarms (hd5 & hd6) and allowing all hop alarms to have a range of -999 to 999 minutes, with the negative range being minutes after the program completes, so if I add flame out hops that I want to sit for 20 minutes I could set hd5 to -20?