matsstaff/stc1000p #96: S0 too powerful for ramping between mash steps

karlp, 2016-08-12:
When I want to heat up strike water, I want to use S0=100 to bring cold tap water up to heat, but after dough in, if I've dropped down to 65C or so, using the full S0 output to get back to PT1 results in major overshoot.  My vessel isn't circulating pumped, so I've got to find some sort of better place to mount the thermometer maybe, but what would really help is having a different factor for the mash ramp output.

matsstaff, 2016-08-12:
This is a duplicate of #93 .
SO parameter is 'dual use' because originally, I wanted to keep the amount of parameters limited, but I guess we're passed that point already... I intend to fix this, but I've been busy with other stuff for the summer.
Still, thanks for your your feedback!

karlp, 2016-08-12:
Yep, absolutely, sorry for not reviewing open issues properly! no rush, just have to learn how to use it.  Still struggling a bit, automation isn't quite hitting my numbers the same way as my old manual temp + alarm clock was :)

closed 2016-08-12 by karlp

matsstaff, 2016-08-12:
No worries :) It is good to know there's interest.
Sorry to hear that, hope it can be sorted out :)
Cheers!