matsstaff/stc1000p #99: looping

MightyRich, 2017-04-27:
Sorry if I posted this in the wrong part, I haven't been using GitHub for long.

I am trying to use these thermostats for my snakes, I need to create a 24 hour loop in a profile and have setpoints where them temp changes to mimic the natural temp variations that happen every day, i can use a profile for each season in the year, this modification is very close to what I am after, I have successfully added this to my thermostat and have set a profile with 6 temps and set points, it worked but only once, I would like the profile to loop, how can I do this please? i also have the problem of getting the thermostat to start at the correct time in the loop?

matsstaff, 2017-04-28:
Hi!
I'm sorry, but this use case is not covered in the firmware. That is, currently there is no way to do loops.
And I'm not sure it is such a good idea. Here's why: The STC is pretty simple, it does not have a real time clock or even crystal clock. It uses the internal RC oscillator for timing, which is pretty inaccurate, it will drift over time. For beer, It does not matter if it is off by a couple of minutes or even hours over a period of a few weeks. I don't know the required accuracy for your application (but it might be ok, since you are willing to change profile a couple of times a year and thereby re-synchronize).

Secondly, the firmware updates state in EEPROM every hour. EEPROM has limited write cycles (in this case a minimum om 10k write cycles). Running continuously, you'd 'wear' out the controller in a year or two (10k hours is ~400 days).

All is not lost though. I have tweaked the firmware for a minute based version, that differs in that minutes are used as time base *and* to mitigate the EEPROM wear issue, only updates EEPROM after advancing to the next step in the profile. You should be able to use that, since each step can be 999 minutes at most or about 16 hours. Using for example 6 hour steps (360 minutes), should give you 4 EEPROM writes per day and thus your controller should last at least 6 years (2500 days). It would also allow you to 'calibrate' your profile to be within +- 1 minute or so per day, if you find it drifting too much.

So finally, what you need is changing to continuously loop the profile, instead of stopping at the end. And that should be easy enough.
I don't normally do 'one off' firmwares, but since what you want is such an easy change (probably takes less time to write than this answer). I'd be willing to make that change and just send it you, if you are interested in trying that out.

Cheers!
//mats

MightyRich, 2017-04-28:
Oh wow thanks that would be great :)

MightyRich, 2017-05-01:
will you need my email address?

matsstaff, 2017-05-01:
Hi!
I put up a new branch. You can find it [here](https://github.com/matsstaff/stc1000p/tree/looped).
[Here](https://raw.githubusercontent.com/matsstaff/stc1000p/looped/picprog_minute_looped.ino) is a direct link to the .ino.
I just made the change and compiled, haven't tested at all (I don't even have my STC available for testing at the moment), but it is a simple change so I think it should work.

Cheers!
//mats

MightyRich, 2017-05-04:
Thank you very much i'll test it and let you know but as you say it should be ok, thanks again :)