matsstaff/stc1000p #100: support for Holtek HT66F40 (Lerway) based models

dt, 2017-08-15:
I just picked up a Lerway STC-1000 on Amazon and discovered, when I cracked it open to solder on a header for programming it, that the board wasn't quite the same as my old one, in particular it doesn't have the same programming header.

[image]

There are a couple empty headers on the main board though, labeled GND, VDD, RST, NTC 1 and NTC 2. A volt meter suggests pig out between the main board and front panel board it pretty similar: same pins are tied to GND and VDD, the one that connects to the "RST" header seems to correspond to what I expected to be nMCLR, the one leading to the NTC2 header corresponds to where I'd expect ICSPDAT. There wasn't a header for the one that I'd expect to correspond to ICSPCLK though, but I tied just sticking a wire to it to see if I could talk to the thing.

It looks like the chip itself is a Holtek HT66F40. 
[image]

Hooking it up, do seem to be able to interact with it somewhat -- I can switch it to low-voltage, I can switch it back (via `d`), but that's about it -- can't read a device ID or anything. I'm guessing this chip isn't supported, but I'm curious if anyone else has run into this chip or has any pointers on adding support for it.

matsstaff, 2017-08-22:
This is a completely different architecture and is not and probably will not be supported by this project. You can search the massive thread on homebrewtalk and see that is has been discussed before. 
It would be cool though, if it could be 'hacked', but my gut feeling is that it is not really feasible.

The problems are (in order of importance):
* The documentation is lacking and in engrish. No specs of the prococol/timings used for flashing.
* No free tools. C compiler, flash programmer et.c. need to be bought from Holtek.

It would not be an impossible task. But hard. And probably cost a bit of cash. And without a free compiler, not really worthwhile.

dt, 2017-08-28:
Thanks for the details!

closed 2017-08-28 by dt