matsstaff/stc1000p #104: Controlling STC1000p with COM over internet

sveco86, 2018-11-01:
I am running v.1.09 with Communication on my STC1000p. I plugged ESP8266 to controller which is reading current state of controller by reading EEPROM (and publishing values to MQTT).
I want also to control the STC by writing directly to EEPROM (changing profiles and running profiles and changing SP in Th mode). 
Unfortunately, even I change (write) values in EEPROM (e.g. profile number to run (0-6) to EEPROM address 123), nothing happens... Profile runs only when I manually enter menu Set->rn and leave from it OR when I restart STC.
Is it possible to implement few changes?
1. Read config every time when EEPROM address 123 is changed and run profile.
2. When STC is in Th mode, every 1 minute read SP (in EEPROM address 114) and run thermostat according to this value?

kokangit, 2018-11-29:
Oh, what a brilliant idea! Please publish your changes, I'd like to control my STC's with ESP8266.

matsstaff, 2018-12-06:
I'm not 100% sure from the description what you are having problems with...

One thing to check is that the EEPROM addresses really are correct. EEPROM layout might have changed in some release and I may have forgotten to update the COM sketch.... I just don't remember since it's been a while :)
I think most changes to EEPROM really *should* take effect (as long as the correct address is used), I've tried to avoid 'caching' the values in variables and reading them from EEPROM instead. 

I'd be glad to try to answer questions :)
Happy hacking!

fredericsureau, 2019-10-06:
I have the exact same use case.

Setting run mode to PrX through communication does not automatically set dh, St and SP values to the first step of the profile.
When using the physical menu, these values are automatically set.
Maybe this is your problem?

As a workaround, I have solved this problem by setting dh, St and SP from the COM client.

No problem on my side when setting SP in "th" mode, temperature control is instantly updated.