matsstaff/stc1000p #105: build error

sveco86, 2018-11-26:
It is not possible to build following firmware using building scripts in src folder:

- celsius_com
- celsius_probe2
- celsius_fo433
- celsius_minute_probe2
- celsius_minute_com
- celsius_ovbsc

e.g. ./build.sh com  gives:
./build.sh com

Building stc1000p_com
sdcc -c -o build/page0_celsius_com.o page0.c --use-non-free -mpic14 -p16f1828 --opt-code-size --no-pcode-opt --stack-size 8 -DCOM
sdcc -c -o build/page1_celsius_com.o page1.c --use-non-free -mpic14 -p16f1828 --opt-code-size --no-pcode-opt --stack-size 8 -DCOM
sdcc -o build/stc1000p_celsius_com build/page0_celsius_com.o build/page1_celsius_com.o --use-non-free -mpic14 -p16f1828 --opt-code-size --no-pcode-opt --stack-size 8 -DCOM
message: Using default linker script "/usr/share/gputils/lkr/16f1828_g.lkr".
error: No target memory available for section ".code".
Makefile:13: recipe for target 'stc1000p_celsius_com' failed
make: *** [stc1000p_celsius_com] Error 1
rm build/page0_celsius_com.o build/page1_celsius_com.o

matsstaff, 2018-11-29:
Looks like the generated code does not fit. It is a tight squeeze.
My guess is you are using a version of sdcc that produces slightly larger code. 
Try changing to an older sdcc version (I think I've tried 3.4.0 and 3.5.0).

Either that, or you've made changes/additions to the code?

sveco86, 2018-11-29:
Hello,
thank you for your reply.

I am using SDCC version 3.8.0, I will try to use older version.

I would like to implement 2 changes described in Issue 104 (https://github.com/matsstaff/stc1000p/issues/104) which will allow me to control STC over internet (ESP8266 using MQTT) by changing SP in Th mode and running profiles.

sveco86, 2018-12-06:
I can confirm that using sdcc 3.3.0 for compilation works.

closed 2018-12-06 by sveco86

matsstaff, 2018-12-06:
Nice to hear you got the compilation working!