matsstaff/stc1000p #108: Change Probe with 10K 3950

tava999, 2020-11-24:
It's possible to change the original NTC 10K 3435 with NTC 10K 3950?
If it's possible, how i can do?
Thanks

matsstaff, 2020-12-01:
Hi!
Sorry for the late reply.
Yes, it probably would be possible. The quick and dirty solution would be to just replace the probe and live with the fact that temperatures will be inaccurate. You could calculate the difference and use a table on a sheet of paper etc to translate. It would still be pretty accurate in a region around 25deg C.

The 'proper' solution is to use the probe specific temperature-resistance table to calculate a new lookup-table for the firmware. I have made a program to make updating the LUT easier here:
https://github.com/matsstaff/ntc-lut-generator
Though, you'd need to set up a development environment and build the project yourself for that.

Last idea is a bit of a hack, but it probably would be possible to locate the lookup-table in the hex data and replace it with your own custom one. 

So, you'd need to put in a bit work to do it properly.

Cheers!
//mats

tava999, 2020-12-01:
Hi mats, thanks for your reply.
I see the LUT generator but i cant run makfile under win10.
I think i should use a Linux OS?
I Need Just to run It and generate the table?
Thank you!
Cheers!

matsstaff, 2020-12-02:
Ok.
The needed steps are:
1. Get the correct datasheet for the probe. You need an accurate temperature/resistance table for that specific probe.
2. Create a text file using that table similar to the one that I prove with the LUT generator for the STC-1000 probe, here: https://github.com/matsstaff/ntc-lut-generator/blob/master/ntc-10k-3435.txt but substitute for actual resistance values for you probe.
3. Download, build and run the LUT generator on your file. Yes, you will need Linux. You can install WSL on Win10 or create a Virtual Machine or ...
4. The last couple of lines of output from the LUT generator will produce the actual code that is needed.
5. Set up the needed tools to build STC-1000+ https://github.com/matsstaff/stc1000p/wiki/2.1-Setting-up-a-development-environment again, Linux is needed.
6. Download STC-1000+ code and replace the LUT's (at rows 31 and 33 respectively -> https://github.com/matsstaff/stc1000p/blob/master/src/page0.c ) with your custom ones.
7. Build STC-1000+, upload to arduino and then flash you unit.

If you don't have much Linux and/or software development experience, you might want ask someone locally to help you out with this.

Cheers!

tava999, 2020-12-04:
Thanks again for the reply.
I managed to create the new look-up table.
Can I just replace it in the code or do I have to necessarily recompile with GPUTILS and SDCC?
Thanks again

matsstaff, 2020-12-04:
Yes, you'll need to recompile (with the changes you've made to the LUT).