matsstaff/stc1000p #109: Off-Grid Solar Refrigerator

mbr1989, 2021-08-11:
Hi,

would it be possible to write a firmware that uses the heating output to switch on a 230V inverter?

best regards
Marcus

mbr1989, 2021-08-11:
Hi,

it is a EPEVER IP Plus 2000 22 (24 Volt) pure sine wave inverter.
https://www.epsolarpv.com/product/72.html

The inverter has a potential-free contact to turn it on.

The fridge thermostat is already disassembled.

Cheers!

mbr1989, 2021-08-11:
https://assets.danfoss.com/documents/35412/AN223886433202en-000101.pdf

https://assets.danfoss.com/documents/35418/AN223886433209en-000101.pdf

... this is to expensive

matsstaff, 2021-08-11:
Ok.Well, if that is the case, I guess it might make sense. That is, it could be do-able.
You'd at least need to modify the stc for 12v operation. Not that hard.
You'd also need to do some firmware hacking. That might be a bit more challenging :)
Basically, I still think it would be easier to just use an existing thermostat and build the delay circuit.

Cheers!

mbr1989, 2021-08-12:
> You'd at least need to modify the stc for 12v operation. Not that hard.

https://www.amazon.de/gp/product/B07RD9DRXZ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Bought the 24 Volt version. 

> Basically, I still think it would be easier to just use an existing thermostat and build the delay circuit.

How the hell is that supposed to work ???

I've already thought about it, but haven't come to a solution

matsstaff, 2021-08-12:
Ok, first off, not all STC-1000's are equal. There have been numerous different 'clones' spotted, all using varying microcontrollers (they look similar from the outside though), so it depends on what actual hardware you got if it can even be reflashed.

Secondly, that fridge thermostat you pictured can most likely be thought of as a temperature actuated mains switch, it probably just sits in series with the mains connection to the compressor, closing the switch (turning on the compressor) when temperature in the fridge rises. It *would* be potential free if you disconnect the spade connectors from it... Note that these thermostats are mechanical devices, the do not *need* mains power to function, they actuate by the change in pressure of a contained gas due to change in temperature.

Here's more info: 
https://assets.danfoss.com/documents/163014/AI357347699370en-000101.pdf
https://store.danfoss.com/en/Climate-Solutions-for-cooling/Appliance-Controls/Standard-thermostat%2C-B62/p/077B6222

My suggestion is, that you disconnect mains in and mains out from the thermostat, short those two leads together (making the compressor always run) and connect the 'potential free contact' on the inverter to the thermostat instead. 
Should this not work as is, *then* look for a relay (or contactor) delay module to delay the start of the fridge until the inverter has had enough time to start.

Cheers!

matsstaff, 2021-08-12:
> The inverter is also for other electricity consumer
See, you didn't mention that before :)
Well, no biggie... Do it this way then... Disconnect the thermostat, feed it 12V instead and let it provide power to two relays. One for 'potential free contact' on the inverter and one for mains for the compressor. 
You could use something like this: 
https://www.ebay.com/itm/124367538472
For both the relays and still have a couple of spares. 
See this for usage:
https://youtu.be/RjWW4nbUyio?t=264
You could do the same thing with an external STC-1000, if you really want to use an external thermostat. 

> Well sounds not much easier to me.
Easier than what? Modifying the firmware of a thermostat that may not even be modifiable?
Let's put it this way. If I wanted to do what you want to do now, I would do it the way I suggested (and I got some known "good" STC's and I'm pretty familiar with writing code for them) :)

Cheers!